slic3r/Slic3r #5148: Add printer / print settings profile to .gcode header comments

smartin015, 2022-10-15:
**Is your feature request related to a problem? Please describe.**

I'm the maintainer of the [Continuous Print Queue](https://github.com/smartin015/continuousprint) plugin for Octoprint, which has features requiring an understanding of the printer for which a gcode file is sliced (see [docs](https://smartin015.github.io/continuousprint/printer-profiles/#automatic-profile-assignment)). In particular, this is necessary for cluster/network printing when the type of printer on the network varies.

I'm intend to [add profile inference for the most popular slicers](https://github.com/smartin015/continuousprint/pull/133), but Slic3r has no way to examine the gcode and infer the printer to print the file with. I can contribute a PR to add this as a file comment (e.g. `; printer_profile = "CR-20"`) but I wanted to file an issue first in case discussion was needed before I got started.

**Describe the solution you'd like**

A string like `; printer_profile = "CR-20"` is present in gcode, matching the name of the printer profile being used to slice the file.

**Describe alternatives you've considered**

Infer a compatible printer name from implicit details (e.g. build volume) but that's very brittle.

**Additional context**

I made a similar request to Cura: https://github.com/Ultimaker/CuraEngine/issues/1747

However, I'm not familiar with perl scripts and need more pointers if I were to help contribute something here.

smartin015, 2022-11-11:
Hey, could I get a weigh-in on making this GCODE header addition so folks can identify the printer targeted by a gcode file? I intend to learn perl for advent of code this year, would be great to put that to use here :)

smartin015, 2022-11-26:
Friendly ping?

xoan, 2022-11-27:
There is a "Notes" field in Printer Settings where anyone can put anithing you want, like:

```
; printer_profile = "Awesome RepRap Printer"
```

I think It should work in the manner you want.

smartin015, 2022-11-27:
Ok, thanks - I suppose a little manual configuration is fine for the (probably small) subset of folks who use CPQ with slic3r gcode flies.

closed 2022-11-27 by smartin015