slic3r/Slic3r #5153: Can't export to multiple SVG's from commandline.

PMKrol, 2022-12-23:
**Describe the bug**
Can't export to multiple SVG's from commandline.

**To Reproduce**
 slic3r  --layer-height 0.1 --export-svg piratecoin.stl 
 slic3r  --layer-height 0.1 --export-svg piratecoin.stl -o pira-mul 
  slic3r  --layer-height 0.1 --export-svg piratecoin.stl -o ./
  slic3r  --layer-height 0.1 --export-svg piratecoin.stl --output-filename-format [input_filename_base]_[layer_num].svg
  Slic3r/slic3r.pl  --layer-height 0.1 --export-svg piratecoin.stl 

**3D Model and Slic3r Configuration Export**
[piratecoin.zip](https://github.com/slic3r/Slic3r/files/10296954/piratecoin.zip)
https://www.thingiverse.com/thing:3950452

**Expected behavior**
piratecoin001.svg ... piratecoin999.svg.

**Desktop (please complete the following information):**
 - OS: Kubuntu 20.04
 - Version 1.3.0
 
**Additional context**
Read some of svg-relateg issues (e.g. https://github.com/slic3r/Slic3r/issues/4928), but didn't figure it out...

arslan437, 2023-01-03:
Did you find the solution to your problem. 

Because I am also looking for the command line tool with which I can save the sliced parts into sperate svg or jpeg fromat so later I can dsiplay those on lcd and cure the resin.

PMKrol, 2023-01-09:
Unfortunately, started to writing my own script.

arslan437, 2023-01-09:
Is it going to be open source?  
Are you going to make your own slicers or a script that takes the single svg and convert it to multiple files ?

PMKrol, 2023-01-11:
It's going to be command line tool to slice stl to svgs and eventually to gcode (maybe by svg2gcode or similar).
Yes, it will be open-sourced.

arslan437, 2023-01-11:
That's great. Did you created the repository? I want to follow it.

arslan437, 2023-01-11:
Also check the prusa slicer is also based on this repository.

PMKrol, 2023-01-11:
I did! Unfortunately, it won't solve my problem.
I'll put it on github when version prealpha-0.001-notworkingyetbutcoolenought will be done, promise ;)

arslan437, 2023-01-11:
In which language you are developing?

PMKrol, 2023-01-11:
As far as I remember, I tried all.
C++. But I'm not a programmer ;)

arslan437, 2023-01-11:
Why do you want to make your own slicer ?