slic3r/Slic3r #5158: Is Slic3r dead?

kevinoue, 2023-03-10:
Download links are down on the home page, is Slic3r Dead?

moltra, 2023-03-21:
I am wondering the same thing.

luzpaz, 2023-04-02:
Still down :disappointed:

elXaptation, 2023-04-19:
+1
Wondering the same. I can't get 1.3.x to build on Arch anymore. AppImages also busted. Awe shucks.. gonna have to use Prusa-Slicer :(

Prusa-Slicer has a SNAP... [shurg]

flishy2532, 2023-05-01:
i found can download Slic3r 1.3.0 in this link http://dl.slic3r.org/win/ 
change https to http

BETLOG, 2023-12-15:
So what's the deal? Did all of the contributors just suddenly become mute? Or has there been some kind of parting statement made?
The files are still available, but it seems that there hasn't been any code contribution for numerous years.
I'm returning to 3dprinting after several years of zero interaction, and finding the slic3r git like this is pretty disturbing:
It's hard to know if slic3r has achieved god status and simply no longer needs to evolve, or if i'm in some kind of Truman Show terrarium, or if all major and minor contributors have died/completely lost interest. Is my simulation glitching?

What are users doing? Switching to prusaslicer or something closely related?

supermerill, 2024-01-16:
The mainline hasn't evolved since prusaslicer took over.
It's not possible to "compete" with prusaslicer which have a full team with only some days of works per month as a side-project.
But slic3r legacy continue in prusalicer, superslicer, bambu and orca.
For the future of this repository and the slic3r brand, it's not decided yet.