slic3r/Slic3r #5172: Slic3r does not open settings

AleyRobotics, 2023-10-24:
**Describe the bug**
Application crashes after launch 

**To Reproduce**
Steps to reproduce the behavior:
1. Copy to applications Repetier-Host-Mac_1_2_1 at MacOS Sonoma (14.0)
2. Start Repetier-Host
3. Open Slic3r settings
4. Slic3rCrash, look MacOS crashlog

**Tried to start from command line:**
1. $ cd /Applications/Repetier-Host\ Mac.app/Contents/Resources/Slic3r.app/Contents/MacOS/
2. $ ./Slic3r
**Command line reports:**
Can't load './local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle' for module Cwd: dlopen(./local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle, 0x0001): tried: './local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle' (relative path not allowed in hardened program), '/System/Volumes/Preboot/Cryptexes/OS./local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle' (no such file), '/usr/lib/./local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle' (no such file, not in dyld cache), './local-lib/lib/perl5/darwin-thread-multi-2level/auto/Cwd/Cwd.bundle' (relative path not allowed in hardened program) at DynaLoader.pm line 197.
 at FindBin.pm line 83.
Compilation failed in require at FindBin.pm line 83.
BEGIN failed--compilation aborted at FindBin.pm line 83.
Compilation failed in require at ./slic3r.pl line 7.
BEGIN failed--compilation aborted at ./slic3r.pl line 7.

**MacOS crashlog**

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               perl-local [885]
Path:                  /Applications/Repetier-Host Mac.app/Contents/Resources/Slic3r.app/Contents/MacOS/perl-local
Identifier:            perl-local
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        Repetier-Host Mac [728]
Responsible:           Repetier-Host Mac [728]
User ID:               501

Date/Time:             2023-10-24 22:19:38.9547 +0300
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        04D30C1F-E7C6-49A3-BA46-CBA7B8D8639B

Time Awake Since Boot: 370 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00007ff7b262efe0
Exception Codes:       0x0000000000000002, 0x00007ff7b262efe0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [885]

VM Region Info: 0x7ff7b262efe0 is in 0x7ff7aee2f000-0x7ff7b262f000;  bytes after start: 58720224  bytes before end: 31
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_MEDIUM            7fcdc8000000-7fcdd0000000 [128.0M] rw-/rwx SM=PRV  
      GAP OF 0x29dee2f000 BYTES
--->  STACK GUARD              7ff7aee2f000-7ff7b262f000 [ 56.0M] ---/rwx SM=NUL  ... for thread 0
      Stack                    7ff7b262f000-7ff7b2e2f000 [ 8192K] rw-/rwx SM=SHM  thread 0

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	    0x7ff812f17116 nanov2_malloc_zero_on_alloc + 75
1   libc++abi.dylib               	    0x7ff8130975aa operator new(unsigned long) + 26
2   libwx_osx_cocoau_core-3.0.dylib	       0x10f0c6164 wxStockGDI::GetFont(wxStockGDI::Item) + 356
3   libwx_osx_cocoau_core-3.0.dylib	       0x10f0c619f wxStockGDI::GetFont(wxStockGDI::Item) + 415
4   libwx_osx_cocoau_core-3.0.dylib	       0x10effcb87 wxSystemSettingsNative::GetFont(wxSystemFont) + 71
5   libwx_osx_cocoau_core-3.0.dylib	       0x10f0c6174 wxStockGDI::GetFont(wxStockGDI::Item) + 372
6   libwx_osx_cocoau_core-3.0.dylib	       0x10f0c619f wxStockGDI::GetFont(wxStockGDI::Item) + 415
7   libwx_osx_cocoau_core-3.0.dylib	       0x10effcb87 wxSystemSettingsNative::GetFont(wxSystemFont) + 71
[... 444 lines cut ...]
      },
      "deploymentId" : 240000012
    },
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 240000021
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000017
    }
  ]
}
}

`

**Desktop (please complete the following information):**
 - OS: MacOS Sonoma 14.0 intel

Chromeking, 2023-11-03:
Any thoughts or updates on this one?  I am experiencing same issues. on Macbook air M1 with 14.1 installed.

Chromeking, 2023-11-03:
> Any thoughts or updates on this one? I am experiencing same issues. on Macbook air M1 with 14.1 installed.

let me ammend that we can't even get the application to load.  it starts and gets no farther than that.

mayaway, 2023-11-14:
M1 Mac Silicon 14.0
App launches then fails to open. Hope this can be made compatible with Sonoma
It's a great little piece of software - have used for years.

ihatethecloud, 2024-01-13:
Same issue on Mba M2

mayaway, 2024-01-13:
I installed Prusa Slicer - free, open source -- and created a custom template for my obsolete printer: Maker Farm 8 inch Pegasus
I am back in business!

arnlen, 2024-05-09:
Same with:
- MBP Intel
- Sonoma 14.4.1

Any progress on this one?

mayaway, 2024-05-09:
I now use PrusaSlicer which works just as well, and you may create your own "custom printer" settings - which I needed to do for my 8" MakerFarm kit which is out of business and obsolete...