slic3r/Slic3r #5180: Slic3r crashes on launch with "PC register does not match crashing frame" - mac Sonoma 14.0

EricPayrot, 2023-12-12:
Hello,

I re-installed slic3r on a new macbook Air G2 under macOs Sonoma 14.0. When launching the app, it crashes immediately (doesn’t even open a window). In the crash report, I get a “PC register does not match crashing frame” error.

I also tried the dev version slic3r-1.3.1-dev-e37620f-osx and get the same issue.

It may not be linked but I initially migrated the app from my old macbook pro, which carried the setting files along. After my issues, I deleted the whole slic3r settings folder in application support folder. But it did not solve the issue.

Any idea would be very appreciated
thanks
Eric

paulmovricnz, 2023-12-23:
I am having the same problem. The error in the error log is "Error Code:      0x00000004 (no mapping for user data read)".

Hope this helps.

AleyRobotics, 2024-01-12:
looks like same bug https://github.com/slic3r/Slic3r/issues/5172