slic3r/Slic3r #5182: Bug Report/Feature Request: Extrusion Calculations At Corners Result In Unsightly Bulges

rf1k-mjh11, 2023-12-31:
This will probably end up as a feature request, as it seems all slicers have been living with this issue 'forever'.  It just seems that no one has noticed it until now.

**Describe the bug**
I believe an issue exists where the extrusion amount is incorrectly calculated at corners.  This error leads to localized over-extrusion which in turn can lead to the bulges at the corners.
If you split a straight extrusion path at any point, you have something like this (when viewed from the top, along hot end axis):
[image]
The way I believe Slic3r calculates the extrusion amount, it doesn’t matter in which direction the second (green) path leads, either straight on (as in the picture) or at an angle, such as this:
[image]
A close look at the indicated area reveals the over-extrusion:
[image]
The calculated extrusion amount is shown in green and you can see the issue.  Material is extruded where material was previously extruded, leaving no room for the new material.  At angles as small as shown above, that over-extrusion issue may go unnoticed.  But at a 90° angle, the situation is more severe:
[image]
The area of excess extruded material is shown shaded dark.

For prints with only one perimeter, the excess material is apt to escape to both sides of the bead, causing a minor bulge.  But if the settings call for multiple perimeters, the first (innermost) perimeter will bulge the same as a single perimeter.  Any bulging to the side of the next perimeter takes up space that will be needed for the next perimeter.  As soon as the subsequent perimeter(s) are printed, the material can no longer escape to both sides and can only bulge **away** from the previous perimeter(s).  This bulge again takes up room needed for the next perimeter, thereby **compounding the problem** with each **additional perimeter**.
The bulge occurs on the face after the corner, as that is the side that leads to the problem – after ‘turning’ the corner.

The internet has a few pictures showing the bulge on both corners of the face, but this may be caused by additional factors (true over-extrusion, loose belts, etc.).  All of my examples here at home show only one bulge per side.  Here is a simulated bulge to clarify the issue:
[image]

I do not know the mathematics needed to calculate the extrusion amount.  
So I used the generated GCode of a model with a 50mm long side and a 1mm long side.  Since I could find no difference in the extruded amounts (one was exactly 50x the other), I can only assume there is no compensation for the over-extrusion error.

**To Reproduce**

  _Not applicable as the issue pertains to the existing calculation model._

**3D Model and Slic3r Configuration Export**

  _Issue exists with all models having vertical corners with angles around 90°_

**Expected behavior**

 N/A

**Screenshots**

See above

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: probably all Slic3r and PrusaSlicer versions (maybe other slicers, too).

**Additional context**
Several websites attempt to blame the corner bulges on jerk, (constant) over-extrusion, printing speed, or anything else they can think of.  I believe it can be attributed to this minor error, especially if compounded by multiple perimeters.

To correct the issue one would have to reduce the amount extruded as long as an overlap exists.  The reduction amount can be calculated accurately.  The question is, do you want to further complicate path and extrusion calculations?  (Maybe a simple 'fudge' reduction-factor that remains in effect for half of the extrusion width, after any corner beyond 60° would be sufficient?)

rf1k-mjh11, 2024-01-20:
The overlapping area (which becomes an extrusion volume when multiplied by the layer height and the extrusion multiplier) can be calculated for any angle as follows:

Known:
**BW**: the bead width
**α**: Included angle (for obtuse angles this is >90°, for acute angles <90°)

Sought:
**OA**: Overlapping area

OA = 2*((BW/2)² * tan(90- α /2)/2 - BW² * π/4/360*(90- α /2))
(I'm not claiming that this is the 'cleanest', most elegant or shortest formula - it was simply derived from looking at the intersection area).

Let's look at an overlapping area. Here is one at a slightly acute angle:
[image]

The area of overlap is outlined in yellow. The included angle α is slightly less than 90°.

If you create a line of symmetry, two congruent right-angled triangles are created:
[image]

Using these triangles, we can calculate the overlap area.

One of the legs of the triangle is formed by half the width of the bead 'BW'. The other leg corresponds to the tangent of the angle (90°- α/2) times half the width of the bead.

The area of a triangle is calculated as 'base times height divided by 2'. As it is a right-angled triangle, the area of the triangle is easy to calculate: 'one leg times the other leg divided by 2'.

This gives you the area of the triangle, from which the area of the 'circular sector' must be subtracted.

The area of the circular sector is calculated by dividing the area of the circle by 360° times the angle, in this case (90°- α/2).

After subtracting the area of the circle sector, one has **exactly half the overlapping area**. The result must therefore **be doubled**. This is how I derived the formula above.

You can use the formula to calculate the intersection area at corners with any angle. However, Slic3r may no longer calculate the paths, as expected, at very acute angles.  (The Arachne Engine definitely does things differently for acute angles starting somewhere around 30°.)

This overlap, with the resulting over-extrusion, only needs to be taken into account over a certain path length.  This path length is, again, 'the tangent of the angle (90°- α/2) times half the width of the bead'.  The actual **rate of compensation versus length** has yet **to be determined**.

There are several posts on this topic on the RF1000.de forum, where the issue was originally discussed, under this thread (unfortunately they are in German):
[originating post]https://www.rf1000.de/viewtopic.php?p=37366&sid=5d5ed4a102f0964977fcfd87d083ac29#p37366

***mjh11***

supermerill, 2024-01-24:
already implemented here:
https://github.com/slic3r/Slic3r/commit/90b129e5857e0f6b93d347497d23f458e4c0a2d1

rf1k-mjh11, 2024-01-24:
Per previous comment, closed as of slic3r 1.3.0.

BTW, is it still experimental?

closed 2024-01-24 by rf1k-mjh11

rf1k-mjh11, 2024-01-24:
Thank you supermerill,

Thanks to you I can see the source code of the implementation.

There is a note stating experimental status:

external_perimeter_cut_corners setting

	it reduce the flow around corners, depending of the angle.
	experimental status! not tested yet!

How do I get to test it?  I just checked the sliced output of /v1.3.0/
and /1.3.1 (dev)/ [Build commit:
05a2f9028a35c56e039cf7a531d63b7ff7e97645] and could not find any corner
compensation implemented in either version.

Thank you,

rf1k-mjh11

rf1k-mjh11, 2024-01-24:
> It was improved in some other commits.
> It's implemented in the 'superslicer' side of slic3r. You can test it
> with superslicer.

Hi supermerill,

I've now installed two versions of SuperSlicer
(SuperSlicer_2.4.58.5_win64_220924 and SuperSlicer_2.5.59.6_win64_231231).

Neither of them show signs of 'corner cutting' having been implemented. 
At least not for the all too common 90° corner.  In the link to the code
you sent me (90b129e
<https://github.com/slic3r/Slic3r/commit/90b129e5857e0f6b93d347497d23f458e4c0a2d1>)
it says:

//extrude in three steps: one short with big mult, one of nozzle size
with the rest and then the normal one.
//it's a very rough approx of a cos.

Based on that I should see each straight path leading from a corner
split into 3 parts.  Two of these parts which should be right at a
corner.  I don't see that here:

Note: the currently highlighted section is a straight 50mm long path. 
This is not split in any way.  And immediately preceding that is a 3mm
straight path that is not split either.

Am I still missing something?  Is there some setting I need to turn on
in SuperSlicer?  (Nice slicer, by the way.)  Or are there prerequisites
that need to be met for the 'corner cutting' to be active?

Sorry for bothering you with this and thank you for your time,

rf1k-mjh11

supermerill, 2024-01-25:
Yes, you have to activate it.
100% value, means it will remove exactly the needed amount (there is threshold value, it won't remove more than half, equivalent to a 35° angle)
You can search settings with the search tool (the magnifying glass)
[image]