slic3r/Slic3r #5187: crush after installing on MacOS Sonata 14.4.1

jiashuow, 2024-04-09:
After installing the slic3r 1.3.0 dmg file from the website and Dropbox, I can't open it when I click the logo of slic3r. It's not because the system security stopped me from doing so. I have fixed that. 

[image]

Is it because the slic3r doesn't support the newest edition of the MacOS system?

AleyRobotics, 2024-04-13:
same bug https://github.com/slic3r/Slic3r/issues/5172

elpollitodiablo, 2024-05-13:
So I just checked this repo, and it seems abandoned, you may want to use [PusaSlicer](https://github.com/prusa3d/PrusaSlicer/releases)

dwillmore, 2024-05-13:
As long as you want to support the project that killed off this project,
sure go do that.  Or, go use SuperSlicer or *any* other slicer.

snoopyonline78, 2024-05-14:
Same problem

SoCuul, 2024-06-12:
> After installing the slic3r 1.3.0 dmg file from the website and Dropbox, I can't open it when I click the logo of slic3r. It's not because the system security stopped me from doing so. I have fixed that.
> <img alt="Screenshot 2024-04-08 at 8 21 49 PM" width="619" src="https://private-user-images.githubusercontent.com/144744464/320682288-4b0f27b1-04dc-4b9c-8068-2c1065248d60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxNjYyOTIsIm5iZiI6MTcxODE2NTk5MiwicGF0aCI6Ii8xNDQ3NDQ0NjQvMzIwNjgyMjg4LTRiMGYyN2IxLTA0ZGMtNGI5Yy04MDY4LTJjMTA2NTI0OGQ2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxMlQwNDE5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OWU5ZDNlNGYwMmUxYzM1ZGI3MjBlZGVkNDJjYzFlYzU0NzVjMjBjMTJhMDBjZGMxOWFlZDY0MTE3ZDNhM2VhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UWnHjiwToxHeJUbu2XMW-4YKtT8M84LcMWHOGMmCt5c">
> 
> Is it because the slic3r doesn't support the newest edition of the MacOS system?
@snoopyonline78 

Slic3r has been abandoned for a long while now. I'd recommend you try https://github.com/SoftFever/OrcaSlicer for the one of the best slicers. It's based on BambuStudio, which in turn is based on Slic3r.