slic3r/Slic3r #5197: Slic3r fails to build against Boost 1.88 due removal of asio::io_service

onitake, 2025-08-11:
**Describe the bug**
`asio::io_service` was removed in Boost 1.66, but it was still available as an alias until Boost 1.87, where it was removed.
The new name is: `asio::io_context`

**To Reproduce**
Steps to reproduce the behavior:
1. Install Boost 1.88
2. Build Slic3r against Boost 1.88
3. Wait for the error:
```
src/libslic3r/GCodeSender.hpp:40:11: error: ‘io_service’ in namespace 
‘Slic3r::asio’ does not name a type; did you mean ‘use_service’?
   40 |     asio::io_service io;
      |           ^~~~~~~~~~
      |           use_service
```

**3D Model and Slic3r Configuration Export**
N/A

**Expected behavior**
Slic3r should build against Boost 1.88.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Debian Linux sid/forky
 - Version 1.3.0

**Additional context**
This problem was discovered during a rebuild of the Debian package of Slic3r against Boost 1.88: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1110676

A similar issue was reported here: https://github.com/chriskohlhoff/asio/issues/1653
Further information and a link to the asio/boost::asio changelog can be found there.

sl1pkn07, 2025-09-13:
need more work than simply change io_service to io_context

```
src/libslic3r/GCodeSender.cpp: In member function ‘bool Slic3r::GCodeSender::connect(std::string, unsigned int)’:
src/libslic3r/GCodeSender.cpp:102:14: error: ‘class boost::asio::io_context’ has no member named ‘post’
  102 |     this->io.post(boost::bind(&GCodeSender::do_read, this));
      |              ^~~~
src/libslic3r/GCodeSender.cpp: In member function ‘void Slic3r::GCodeSender::disconnect()’:
src/libslic3r/GCodeSender.cpp:162:14: error: ‘class boost::asio::io_context’ has no member named ‘post’
  162 |     this->io.post(boost::bind(&GCodeSender::do_close, this));
      |              ^~~~
src/libslic3r/GCodeSender.cpp:164:14: error: ‘class boost::asio::io_context’ has no member named ‘reset’; did you mean ‘restart’?
  164 |     this->io.reset();
      |              ^~~~~
      |              restart
src/libslic3r/GCodeSender.cpp: In member function ‘void Slic3r::GCodeSender::send()’:
src/libslic3r/GCodeSender.cpp:447:14: error: ‘class boost::asio::io_context’ has no member named ‘post’
  447 |     this->io.post(boost::bind(&GCodeSender::do_send, this));
      |              ^~~~
```

exist patch for this?

greetings

sl1pkn07, 2025-09-13:
maybe like this?

~~~patch
diff --git a/xs/src/libslic3r/GCodeSender.cpp b/xs/src/libslic3r/GCodeSender.cpp
index 5b536031b..556fde109 100644
--- a/xs/src/libslic3r/GCodeSender.cpp
+++ b/xs/src/libslic3r/GCodeSender.cpp
@@ -97,12 +97,12 @@ GCodeSender::connect(std::string devname, unsigned int baud_rate)
     fs.open("serial.txt", std::fstream::out | std::fstream::trunc);
 #endif
     
-    // this gives some work to the io_service before it is started
+    // this gives some work to the io_context before it is started
     // (post() runs the supplied function in its thread)
-    this->io.post(boost::bind(&GCodeSender::do_read, this));
+    boost::asio::post(this->io, [this]() { this->do_read(); });
     
     // start reading in the background thread
-    boost::thread t(boost::bind(&asio::io_service::run, &this->io));
+    boost::thread t(boost::bind(&asio::io_context::run, &this->io));
     this->background_thread.swap(t);
     
     // always send a M105 to check for connection because firmware might be silent on connect 
@@ -159,9 +159,9 @@ GCodeSender::disconnect()
     if (!this->open) return;
     this->open = false;
     this->connected = false;
-    this->io.post(boost::bind(&GCodeSender::do_close, this));
+    boost::asio::post(this->io, [this]() { this->do_read(); });
     this->background_thread.join();
-    this->io.reset();
+    this->io.restart();
     /*
     if (this->error_status()) {
         throw(boost::system::system_error(boost::system::error_code(),
@@ -444,7 +444,7 @@ GCodeSender::send(const std::string &line, bool priority)
 void
 GCodeSender::send()
 {
-    this->io.post(boost::bind(&GCodeSender::do_send, this));
+    boost::asio::post(this->io, [this]() { this->do_send(); });
 }
 
 void
diff --git a/xs/src/libslic3r/GCodeSender.hpp b/xs/src/libslic3r/GCodeSender.hpp
index 8f61a80b2..5b730f31e 100644
--- a/xs/src/libslic3r/GCodeSender.hpp
+++ b/xs/src/libslic3r/GCodeSender.hpp
@@ -47,7 +47,7 @@ class GCodeSender : private boost::noncopyable {
     void reset();
     
     private:
-    asio::io_service io;
+    asio::io_context io;
     asio::serial_port serial;
     boost::thread background_thread;
     boost::asio::streambuf read_buffer, write_buffer;
~~~

Woogie-Maker, 2025-11-13:
Just out of interest, why do people still use Slic3r when development seems to have ceased many years ago?

Thanks.